ArekKubacki/Hoymiles-Plant-DTU-Pro #2: Pomoc w uruchomieniu.

duma93, 2022-04-14:
Witaj,

Trafiłem na twój projekt i bardzo by mi się on przydał ale niestety doszedłem do kroku gdzie wgrałem do HACS twoje repotytorium i nieststy nie wie jak dalej zrobić konfiguracje. 

Mógłbym prosic o pomoc ? Co zrobić dalej ? 

Pozdrawiam,
Wojciech Ukleja

ArekKubacki, 2022-04-16:
Dzień dobry,
Czy kiedykolwiek dodawałeś coś do pliku configuration.yaml? Należy dodać coś podobnego jak w moim przykładzie. 
Podajesz adres IP na jakim masz DTU, elementy jakie chcesz śledzić oraz liczbę paneli.
Muszę znać więcej szczegółów aby pomóc. 

Good morning,
Have you ever added anything to your configuration.yaml file? You need to add something similar to my example.
You enter the DTU IP address, items you want to track and the number of panels.
I need to know more details to help.

duma93, 2022-04-29:
Witaj,
próbowałem dodać jak w przykładzie podając swoje IP DTU. i zmieniłem liczbę paneli niestety po zapisaniu sprawdzajac konfigurację wyrzuca błąd "Platform error sensor.hoymiles_dtu - Integration 'hoymiles_dtu' not found."  jeśli byłoby łatwiej możemy kontaktować sie mailowo : duma93@gmail.com

mako777, 2022-05-01:
Cześć,
Mi to wygląda jakbyś wgrał repozytorium ale nie dodał samej integracji z niego, czy jak wejdziesz w HA do zakładki HACS to widnieje tam ta integracja ?

ArekKubacki, 2022-05-02:
Wyrzuć z konfiguracji wpis. Zrób restart. Dodaj do konfiguracji wpis. Zrób restart.
Jeżeli dodatek nie jest wczytany do systemu to wyrzuci błąd przy sprawdzaniu składni, że taki nie istnieje.

Maximus1022012, 2022-06-08:
Mi nie działa mam Dtu-pro S . dostaje taki blad:

Logger: homeassistant.components.sensor
Source: custom_components/hoymiles_dtu/sensor.py:60
Integration: Sensor (documentation, issues)
First occurred: 17:28:43 (1 occurrences)
Last logged: 17:28:43

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 60, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

Ponowne instalowanie i wpisy w configuration,yaml nic nie daly. Przed altualizacją Core tworzyło nowe encje ale nie pobierało danych z Dtu teraz nie widac encji.
 Home Assistant Core 2022.6.4
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1

Maximus1022012, 2022-06-08:
wpis do config
sensor:
   - platform: hoymiles_dtu
     host: 192.168.0.118
  ]  name: Hoymiles PV
     monitored_conditions:
       - 'pv_power'
       - 'today_production'
       - 'total_production'
     monitored_conditions_pv:
       - 'pv_power'
       - 'today_production'
       - 'total_production'
       - 'pv_voltage'
       - 'pv_current'
       - 'grid_voltage'
       - 'temperature'
     panels: 6

ArekKubacki, 2022-06-08:
Moduł nie może połączyć się z DTU. DTU podpięte przez  Ethernet?

dzidziaka, 2022-06-22:
Moje dtu jest dziwne.
Nówka wyciągnięta z pudełka nie wystawiała panelu konfiguracyjnego po wifi. Po resecie też nie.
Dtu skonfigurowane w Hoymiles Cloud (w miejsce starego) wysyła dane. Ale już dwa razy przestało, pomaga wyłączenie/włączenie.
Integracja z HA zadziałała za pierwszym razem, potem po restarcie dtu już nie działa.
Port 502 na dtu jest otwarty, telnetuję się na niego z HA.
Upgrade dtu do V00.02.14 nic nie zmieniło.
Masz Arku jakieś pomysły?

K2M7L, 2022-06-28:
> Moduł nie może połączyć się z DTU. DTU podpięte przez Ethernet?

u mnie to samo. Mam 2 DTU Pro, podłączone po wifi oraz po Ethernet

mój błąd:
Logger: homeassistant.components.sensor
Source: custom_components/hoymiles_dtu/sensor.py:60
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 11:42:24 (1 occurrences)
Last logged: 11:42:24

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 60, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

mój config.yaml (pokazuje wszystkie sensory jakie mam):
sensor:
  - platform: rflink
    automatic_add: true
    wait_for_ack: false
  - platform: hoymiles_dtu
    host: 192.168.0.194
    name: Hoymiles_PV
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
    panels: 10
    scan_interval: 00:02:00

Larson0815, 2022-06-28:
Hello all,

I also would like to follow this topic and discussions. Is it possible to stick to english please!?
Or maybe two languages!?

Thank you!

dzidziaka, 2022-06-28:
It is interesting that with  hm-dtu-pro-modbus-python dtu is show the data. Sometimes from one inverter, sometimes from two or three. Sometimes no data with "problem reading registers":
inverter number:  3
PVVoltage:        36.4 V
PVCurrent:        0.03 A
PVPower:          14.1 W
PVTodayProd:      1009 Wh
PVTotalProd:      36941 Wh
GridVoltage:      229.6 V
GridFreq:         50.0 Hz
Temp:             42.0 oC
OperatingStatus:  3
AlarmCode:        0
AlarmCount:       1
LinkStatus:       263
inverter number:  4
PVVoltage:        29.9 V
PVCurrent:        0.05 A
PVPower:          16.6 W
PVTodayProd:      1402 Wh
PVTotalProd:      38553 Wh
GridVoltage:      229.4 V
GridFreq:         50.0 Hz
Temp:             44.0 oC
OperatingStatus:  3
AlarmCode:        0
AlarmCount:       8
LinkStatus:       263
inverter number:  5
problem reading registers

So, there is a communication.
I have dtu with V00.02.14 firmware.

ArekKubacki, 2022-06-28:
The DTU in the config must have the IP assigned by the router for ETH, not Wifi.
I had huge connectivity issues in V00.02.14.
I have downgraded to V00.02.10.
Support Hoymiles helped me with that. Yes, if there is no connection, e.g. with one microinverter, the connection via modbus is unstable. Stability was better on V00.02.10. 

There will be some changes in the new version, which I test. But I do not have much influence on the modbus communication with DTU.

wwk-github, 2022-06-28:
> I have dtu with V00.02.14 firmware.

DTU PRO?
I also have DTU PRO but I do not see any option to upgrade firmware up to .14.
I have:

Hardware Ver.
H09.02.02

Software Ver.
V00.02.10

How did you manage to get the .14 software version?

dzidziaka, 2022-06-28:
DTU PRO.
Hardware ver.: H09.02.02
My colleague (who have installer account and created my plant in Hoymiles Cloud) upgraded my dtu to .14. I had .12 if I remember well.
I heard that .15 is coming...:)

Larson0815, 2022-06-28:
I have also an DTU Pro but at the Moment only 1 inverter online. Hopefully during the next couple of days I can bring the other inverters (2 more) online.

I know from some supplier that they give like an installer account to the customer because the customer account is sooo limited....
I even don't know FW version of my DTU...

dzidziaka, 2022-06-28:
Arek, what do you think of use dtu's RS485 for get more stable of modbus communictation?
Or that make no sense?

wwk-github, 2022-06-28:
> DTU PRO. Hardware ver.: H09.02.02 My colleague (who have installer account and created my plant in Hoymiles Cloud) upgraded my dtu to .14. I had .12 if I remember well. I heard that .15 is coming...:)

I also have installer-role but I do not see any way to upgrade to .14 
[image]

0A in hex = 10
[image]

Is there any chance to discuss this with your colleague? Maybe he would have an idea?

Larson0815, 2022-06-28:
BY the way, is it planned to get these informations to in the futuer or are they already available:
'operating_status', 
'alarm_code', 
'alarm_count'

ArekKubacki, 2022-06-28:
> BY the way, is it planned to get these informations to in the futuer or are they already available: 'operating_status', 'alarm_code', 'alarm_count'

In the firmware version, these parameters have never changed for me. I think this is a bug in the firmware. I have no way to check, so I'm not planning at the moment. It is currently solving other problems in the module :)

> Arek, what do you think of use dtu's RS485 for get more stable of modbus communictation? Or that make no sense?

I heard that the RS485 interface is more stable. I have to rewrite a large part of the module to do this. I am currently checking it as a separate topic.

wwk-github, 2022-06-28:
> I had huge connectivity issues in V00.02.14.

I believe that might be the reason why .14 is not available. Hoymiles probably rolled this back.

wwk-github, 2022-06-28:
@ArekKubacki is there any simple script existing that could be used to harvest the data and put into influxdb (or any time-series db)?
Something like: https://github.com/sneerix/hm-dtu-pro-modbus-python

ArekKubacki, 2022-06-28:
> 

If it does not require a lot of code, I will try to add it in the next version.

ArekKubacki, 2022-06-28:
> Don't know if this information is helpful or changing anything but I know that other addon with mqtt using these values already.
> 
> But I can somehow understand that it has no priority :-)

If it does not require a lot of code, I will try to add it in the next version.

K2M7L, 2022-06-28:
> Support Hoymiles helped me with that. Yes, if there is no connection, e.g. with one microinverter, the connection via modbus is unstable. Stability was better on V00.02.10.

could you share who to contact and how to define the request what is supposed to be updated? thanks

Larson0815, 2022-06-28:
> 

Thank you, appreciated!

ahowito, 2022-10-20:
Witam, jestem mocno początkujący i walczę z dodatkiem już trzeci wieczór. Proszę o pomoc: gdzie mam to "Ten przykład wymaga niestandardowego:formularza karty kreskowej HACS" wkleić?

ahowito, 2022-10-20:
Witam, jestem początkujący i nie umiem po angielsku. Trzeci dzień próbuję połączyć się z moim DTU pro V00.02.07 i na razie utknąłem na prze klejeniu: "This example needs custom:bar-card form HACS". proszę o pomoc

ArekKubacki, 2022-10-25:
Dodatek jest potrzebny tylko jeśli chcesz takie wykresy jak na screenie. W HACS wystarczy zainstalować dodatek bar-card w zakładce interfejs użytkownika.

ArekKubacki, 2023-02-17:
I have two questions.
1. Did the integration connect to DTU and entities exist in HA?
2. The entity does not exist because you need to enter the name from the serial number of the inverter. Check what entities the integration has created for you and adjust them accordingly in the code.

alekdebski, 2023-02-18:
@ArekKubacki to answer the questions:

1. Yes, as it for example shows correct value of total production. It shows incorrect value of todays production and incorrect value of pv_power. Entities do exist in HA.

2. I've also added ID of microinverters into the dashboard. It started showing some random values - and that's the current problem. To summarize - only total production shows correct value, where other ones show some way too low value, right now my production on app is 2.3kW, HA integration shows:

Looking at the graph what it roughly seems is that it fetched data once and stopped following with requests (it's same on all graphs):
![pv1](https://user-images.githubusercontent
[image]
.com/125669846/219857995-601efc2c-a454-46a4-a2a5-25cc78d9257e.png)

alekdebski, 2023-02-18:
Also - just as a debugging process, I've added integration on clean instance and there is the same issue, it updates values only once and that's it - it stucks on original values.

ArekKubacki, 2023-02-18:
Which firmware version?

PeterAUT, 2023-04-23:
@ArekKubacki

I have also problems to get sensor readings in home assistant.

Installed everthing according your information and added following code in sensor.yaml file:

#DTU PRO
  - platform: hoymiles_dtu
    host: 192.168.178.46
    name: Hoymiles PV
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: 10

DTU Pro is directly connected to router via ethernet. Running 2x HM1500 and 1x HM800 with 10 panels

Hope we can find my fault. I´m pretty new to HA and have zero coding experience.

Is it necessary to enter the micro inverter serial number?

Regards,
Peter

ArekKubacki, 2023-04-24:
Please send logo from HA.

PeterAUT, 2023-04-24:
Hey Arec,

what is the logo? You mean the code?

File: sensor.yaml 

[image]

ArekKubacki, 2023-04-24:
Sorry, logs from HA.

PeterAUT, 2023-04-24:
There is one warning in the log:

2023-04-23 20:08:33.707 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform hoymiles_dtu is taking over 10 seconds.

ArekKubacki, 2023-04-24:
DTU is by ETH? 
Can you ping DTU? 
DTU firmware version? 
HA version?

PeterAUT, 2023-04-24:
I will send you the required information in the evening

PeterAUT, 2023-04-24:
Suddenly I have seen, that dashboard shows some data. The value for actual power, energy today and energy total are correct (energy total is lower in s-miles cloud because microinverters are running longer then dtu pro)
[image]

[image]

Only the single sensors are not available

ArekKubacki, 2023-04-24:
You need to rename the entities on the card. The ones in the example do not have serial numbers. You have to match them with your numbers.

Maximus1022012, 2023-05-10:
Hello, what could be wrong that the values ​​today decrease after 20

[image]

ArekKubacki, 2023-05-10:
This is how DTU works. In the evening, when the inverters turn off, it happens that not all of them will send the last frame. My module only fetches data from the DTU. The total variable is protected against this and is best used for the Energy tab.

Maximus1022012, 2023-05-13:
Hi
[image]
[image]

The total values ​​are consistent with home assistant and Hoymiles, but after 20.00 the discrepancy even as if some inverter did not send the frame. How can you block it because the readings until 20.00 are false

Regards

ArekKubacki, 2023-05-13:
The answer is the same as before. Such data is provided by DTU. Maybe someday I'll protect the variable today. You can try it yourself with templates.

sanelito, 2023-05-25:
Using DTU Pro S

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 304, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 82, in setup_platform
    dev.append(HoymilesPVSensor(name, updater.data.microinverter_data[i-1].serial_number, i, updater.data.microinverter_data[i-1].port_number, variable, updater))
IndexError: list index out of range

FW from DTU is V15.02.23

Any help would be aprreciated.

markopill, 2023-10-24:
thanks great project, got it working!
is it possible in future to add grid consumpion data? or is it not open via modbus?

ArekKubacki, 2023-10-24:
The module is currently downloading all available data from the inverters. To measure such values, you must use an additional meter.

MariuszSzurgot, 2023-11-09:
witam mam problem z uruchomieniem integracji. 
o ile pojawia mi sie ona w HACS nie mam opcji "uruchom" i teraz pytanie czy powinna byc ta opcja. Czy KOD podany w instrukcji wklejam w "/config/configuration.yaml" i czy powinienem tez wkleić tam KOD "Dashboard example"?

bo do config.yaml dodalem tylko konfiguracje i dodajac urzadzenie do dashboardu nic mi sie nie wyswietla. DTU PRO S polaczone przewodowo, IP uzupelnione prawidlowo.

ArekKubacki, 2023-11-09:
> o ile pojawia mi sie ona w HACS nie mam opcji "uruchom" i teraz pytanie czy powinna byc ta opcja.
Nie

> Czy KOD podany w instrukcji wklejam w "/config/configuration.yaml"
Tak

>  i czy powinienem tez wkleić tam KOD "Dashboard example"?
Zdecydowanie nie. Kod wklejasz na wybraną kartę.

> bo do config.yaml dodalem tylko konfiguracje i dodajac urzadzenie do dashboardu nic mi sie nie wyswietla. DTU PRO S polaczone przewodowo, IP uzupelnione prawidlowo.

Tutaj mogą pomóc Logi z HA albo kryształowa kula :)

MariuszSzurgot, 2023-11-09:
> Nie
>
>
> Tak
>
>
> Zdecydowanie nie. Kod wklejasz na wybraną kartę.
>
>
> Tutaj mogą pomóc Logi z HA albo kryształowa kula :)

Pomoesz jak wycignac logi? poczatkuje w HA a nieukrywam przydala by mi sie ta funkcjnalnosc hoymiles.

ArekKubacki, 2023-11-09:
Ustawienia -> lupka ->logi

biasowicka, 2024-02-01:
Witam, czy to normalne, że bardzo często występuje problem z niedostępnymi encjami? Korzystam od jakiegoś czasu, działał różnie ale działało.

Teraz mam problem, bo encje są niedostępne, pomimo tego, że konfiguracja pozostała bez zmian.

HA jest na bierząco aktualizowany oto ostanie LOGI:

Rejestrator: homeassistant.components.sensor
Źródło: helpers/deprecation.py:205 Integracja: Sensor Pierwsze wystąpienie: 31 stycznia 2024 22:07:07 ( 6 wystąpień) Ostatnie zalogowanie: 31 stycznia 2024 22:07:07

Z hoymiles_dtu użyto STATE_CLASS_MEASUREMENT. Jest to przestarzała stała, która zostanie usunięta w HA Core 2025.1. Zamiast tego użyj SensorStateClass.MEASUREMENT, zgłoś to autorowi niestandardowej integracji „hoymiles_dtu”
Z hoymiles_dtu użyto STATE_CLASS_TOTAL. Jest to przestarzała stała, która zostanie usunięta w HA Core 2025.1. Zamiast tego użyj SensorStateClass.TOTAL, zgłoś to autorowi niestandardowej integracji „hoymiles_dtu”
Z hoymiles_dtu użyto STATE_CLASS_TOTAL_INCREASING. Jest to przestarzała stała, która zostanie usunięta w HA Core 2025.1. Zamiast tego użyj SensorStateClass.TOTAL_INCREASING, zgłoś to autorowi niestandardowej integracji „hoymiles_dtu”

----

Rejestrator: pymodbus.logging
Źródło: custom_components/hoymiles_dtu/hoymiles/client.py:98 Integracja: hoymiles_dtu Pierwsze wystąpienie: 31 stycznia 2024 22:07:53 ( 2 wystąpienia) Ostatnie zalogowanie: 31 stycznia 2024 22:07:58

Połączenie z (192.168.8.4, 502) nie powiodło się: przekroczono limit czasu

-----

Rejestrator: homeassistant.components.sensor
Źródło: helpers/entity_platform.py:360 Integracja: Sensor Pierwsze wystąpienie: 31 stycznia 2024 22:07:58 ( 1 wystąpienia) Ostatnie zalogowanie: 31 stycznia 2024 22:07:58

Błąd podczas konfigurowania platformy hoymiles_dtu dla czujnika
Traceback (ostatnie połączenie ostatnie):
  Plik „/usr/src/homeassistant/homeassistant/helpers/entity_platform.py”, wiersz 360, w _async_setup_platform
    czekaj na asyncio.shield(zadanie)
  Plik „/usr/local/lib/python3.11/concurrent/futures/thread.py”, wiersz 58, w ​​trakcie działania
    wynik = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  Plik „/config/custom_components/hoymiles_dtu/sensor.py”, wiersz 69, w setup_platform
    podnieś wyjątek („Nieprawidłowa konfiguracja platformy Hoymiles DTU”)
Wyjątek: Nieprawidłowa konfiguracja dla platformy Hoymiles DTU

-----
Logger: homeassistant
Source: custom_components/hoymiles_dtu/sensor.py:69
Integration: hoymiles_dtu ([documentation](https://github.com/ArekKubacki/Hoymiles-Plant-DTU-Pro))
First occurred: 10:12:54 (1 occurrences)
Last logged: 10:12:54

Error doing job: Future exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform
Logger: homeassistant.components.sensor
Source: helpers/deprecation.py:205
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 10:11:44 (6 occurrences)
Last logged: 10:11:44

STATE_CLASS_MEASUREMENT was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please report it to the author of the 'hoymiles_dtu' custom integration
STATE_CLASS_TOTAL was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL instead, please report it to the author of the 'hoymiles_dtu' custom integration
STATE_CLASS_TOTAL_INCREASING was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please report it to the author of the 'hoymiles_dtu' custom integration

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:408
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 10:12:45 (1 occurrences)
Last logged: 10:12:45

Setup of platform hoymiles_dtu is taking longer than 60 seconds. Startup will proceed without waiting any longer.

Dziękuje za pomoc

ArekKubacki, 2024-02-02:
Tak jest to naturalne, bo DTU nie ogarnia modbusa. Jeżeli DTU nie odpowie na zapytanie no to nowe dane nie przyjdą i tak się dzieje. OpenDTU za kilka złotych i nie ma problemu.

ArekKubacki, 2024-02-02:
https://github.com/ArekKubacki/OpenDTU

Moja wersja pasująca do modułu

biasowicka, 2024-02-02:
Dziękuje, mam problem bo jedną pytkę gdzieś zapodziałem, więc raczej nie będę się bawił i zamówię gotowy zestaw na ebay za 40e wraz z obudową, gdzie już wszystko zmontowane.

Pytanie mam coś takiego jeszcze 

[](https://www.cdebyte.com/products/E49-900MBL-01
)

[image]

Czy na tym da radę co kolwiek zainstalowac i uruchomić ?

ArekKubacki, 2024-02-02:
Esp to raczej nie jest.  Możesz podać link do produktu?

biasowicka, 2024-02-02:
Link

https://www.cdebyte.com/products/E49-900MBL-01

marurb22, 2024-03-10:
Dzień dobry, na karcie mam opis ze encja nie dostępna (dotyczy każdej encji)
Arku czy mogę poprosić o wskazówkę co skopałem 
Poniżej logi z HA
Rejestrator: homeassistant.components.sensor
Źródło: helpers/entity_platform.py:350
integracja: Sensor (dokumentacja, Problemy)
Pierwsze zdarzenie: 20:27:25 (1 zdarzenia)
Ostatnio zalogowany: 20:27:25

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 350, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

biasowicka, 2024-03-10:
może po prostu poczekaj, aż słońce zaświeci :D ?

[19 more replies]