ArekKubacki/Hoymiles-Plant-DTU-Pro #4: Integration does not start

Marcjeok, 2022-10-23:
Hi,
I hav installed this integration a few days ago, but it doesn't seems to work. A few weeks ago the solar panels and hoymiles DTU-pro is installed and I like to add the output to my Home Assistant. The sensors do not appear in Home Assistant. I could really use some help or tips. Did I something wrong? Did I miss something? 
Thanks in Advance!

**What did I do:**
1. Installed HACS
2. Installed this add-on (Hoymiles-Plant-DTU-Pro) as described.
3. Modified my configuration.yaml (see below at Configuration.yaml)

**Configuration.yaml:**
sensor:
  - platform: hoymiles_dtu
    host: 192.168.178.108
    name: Hoymiles PV
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: 6
    scan_interval: 00:02:00

**Logging**
In my logging (Home Assistant) I found the following errors:
Logger: homeassistant.components.sensor
Source: custom_components/hoymiles_dtu/sensor.py:71
Integration: Sensor (documentation, issues)
First occurred: 14:13:42 (1 occurrences)
Last logged: 14:13:42

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 71, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

**Full logs:**
2022-10-23 14:13:09.169 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-10-23 14:13:09.176 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hoymiles_dtu which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-10-23 14:13:42.895 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
await asyncio.shield(task)
File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/hoymiles_dtu/sensor.py", line 71, in setup_platform
raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

ArekKubacki, 2022-10-24:
For the integration to work properly, DTU must connect via the ETH cable. Unfortunately, modbus does not work on Wifi. The logs show that the HA cannot connect to the modbus on the DTU. Are you sure the IP is correct? Is DTU connected via ETH cable?

Marcjeok, 2022-10-24:
DTU is connected via an ETH-cable. WIFI is not configured. I am sure my IP address is correct (assigned with a fixed IP address).

Some (extra) basic Information about the powerplant:
Hardware Ver.  H09.02.02
Software Ver.  V00.02.13

ArekKubacki, 2022-10-25:
Unfortunately, DTU does not always respond to the modbus the first time. Therefore, it is worth restarting the HA several times and checking if it is connected. I also did not check the add-on on your firmware version. Personally, I use V00.02.10 because it is the most stable.

Marcjeok, 2022-11-06:
First of all, thanks for your replies and time!

Here an update about the progress:
Unfortunately, after a few weeks of multiple reboots, nothing happened. After I had installed the Home Assistant update (2022.11.1) last Friday, the sensors of this integration suddenly appeared.
Unfortunately, the sensors of this integration have disappeared by themselves today. After a reboot the log shows the following message (again):

_Logger: homeassistant.components.sensor
Source: custom_components/hoymiles_dtu/sensor.py:71
Integration: Sensor (documentation, issues)
First occurred: 11:37:40 (1 occurrences)
Last logged: 11:37:40

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 71, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform_

What I don't quite understand is why the sensors disappear again if no connection is found. Why don't the sensors continue to exist and are they (temporarily) not updated?

I have asked my installer of the system for help and they made a support call at Hoymiles. Fingers crossed.

ArekKubacki, 2022-11-06:
> First of all, thanks for your replies and time!
> 
> Here an update about the progress: Unfortunately, after a few weeks of multiple reboots, nothing happened. After I had installed the Home Assistant update (2022.11.1) last Friday, the sensors of this integration suddenly appeared. 

DTU is a very unstable device. If there are entities, it means that modbus worked and connected with DTU.

> Unfortunately, the sensors of this integration have disappeared by themselves today. After a reboot the log shows the >following message (again):
>
> __Logger: homeassistant.components.sensor Source: [...]

This is how the add-on works. Variables are if everything is linked.

> I have asked my installer of the system for help and they made a support call at Hoymiles. Fingers crossed.

I have written for support many times. I don't know how they test it all but when they updated the firmware to .14, nothing worked. Go back to version .10 and it works a lot better.

Marcjeok, 2022-11-26:
Hi,

Sorry for my delayed response. I tried to ask for support form Hoymiles, but they Refer me to the installer. My installer has little to no experience with this system, so no real support possible unfortunately.

I started looking for an alternative solution. Since I have all my automations in Node Red, that was my preference. With the help of the internet I have now succeeded (I think). I'm still monitoring everything to see if it really works well. At least, it looks promising!

What I did notice is that if you send multiple Modbus requests in quick succession, the chance is really small (closer to impossible) that there will be a (complete) answer to this. If you send the requests one by one with an interval of, for example, 20 seconds, the result will be a lot better! 
With the help of helpers I have created some fields in Home Assistant and these are now updated twice every 15 minutes. If Modbus does not respond properly, the information will lag a bit, but so far everything is going well.
Maybe with the help of this experience you can do something with your solution, so that others can also be helped with it? Unfortunately I don't have the skills to help or support you with that.

Last but not least, thank you for your integration and all the time and effort you put into it! Keep up the good work ;)

closed 2022-11-26 by Marcjeok