ArekKubacki/Hoymiles-Plant-DTU-Pro #6: Zawieszenie DTU

kmarek1, 2022-12-01:
Mam problem z zawieszaniem DTU .
Gdy włączony jest dodatek Hoymiles-Plant-DTU-Pro zawesza mi się DTU
W przypadku gdy połączenie jest po łan zawiesza się chmura .
Gdy mam połączenie po wi-fi  i po łan zawiesza się modbus TCP .

Hardware Ver.H09.01.02
--
Software Ver.V00.02.07

ArekKubacki, 2022-12-03:
Plugin pracuje stabilnie, niestety DTU już nie. W planach mam przepisać kod i dostosować go aby również działał na RS485. Ale to muszę mieć trochę czasu. Może bliżej końca roku ale nic nie obiecuje.

kmarek1, 2022-12-03:
Co usunąć z pliku /config/custom_components/hoymiles_dtu/sensor.py by otrzymać tylko te encje
- 'pv_power'
- 'today_production'
- 'total_production'
- 'alarm_flag'
?
Mam 20 paneli i mniejsza liczba zapytań modbustcp i chmury być może spowoduje stabilniejszą pracę DTU

ArekKubacki, 2022-12-03:
Tutaj definiujesz, które parametry mają być odczytane przez HA. Moduł czyta i tak wszystkie więc zmiana nic nie da. Zwiększ czas co jaki czyta dane, np. co 5 min.

kmarek1, 2022-12-03:
ustawiłem DEFAULT_SCAN_INTERVAL = timedelta(minutes=10)
Nic to nie dało.
Pytałem o 
> Co usunąć z pliku /config/custom_components/hoymiles_dtu/sensor.py by otrzymać tylko te encje
> 
> 'pv_power'
> 'today_production'
> 'total_production'
> 'alarm_flag'

Fazik83, 2022-12-27:
Jak zmienić czas odświeżania z 2 minut na 1 minutę?

ArekKubacki, 2022-12-31:
W yamlu musisz dodać pod ustawieniami modułu.
```
    scan_interval:
      hours: 1
      minutes: 2
      seconds: 3

```
Oczywiście czasy ustawić pod siebie.

Fazik83, 2022-12-31:
Mam tak

sensor:
  - platform: hoymiles_dtu
    host: 192.168.*.*
    name: Hoymiles PV
    scan_interval: 30
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: *

ArekKubacki, 2023-01-02:
No ale tutaj nawet nie ma tego co podałem wcześniej. Wystarczy kopij wklej.

```
  - platform: hoymiles_dtu
    host: 192.168.*.*
    name: Hoymiles PV
    scan_interval: 30
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: *
    scan_interval:
      hours: 1
      minutes: 2
      seconds: 3
```

closed 2023-04-06 by ArekKubacki