ArekKubacki/Hoymiles-Plant-DTU-Pro #7: Problem po aktualizacji HA Core do 2023.2

januszwoj, 2023-02-02:
Dzisiaj uaktualniłem HA CORE do wersji 2023.2.
Po restarcie przestały mi działać wszystkie urządzenia podłączone po Modbusie - niestety nie pamiętam już komunikatu błędu.
Mam liczniki i przekaźniki oraz bramkę DTU-PRO.
Wyczytałem w googlu że może to być związane z tym że ta wersja HA CORE korzysta z nowszej wersji pymodbus powyżej 3.0.
Wyłączałem po kolei wszystkie urządzenia/moduły i wyszło, że po wyłączeniu Twojego dodatku cała reszta zaczęła działać.
Teraz jak próbuję odblokować konfigurację w pliku sensor.yaml dostaję taki komunikat błędu w "Developer tools"=>"YAML"=>"CHECK CONFIGURATION"

Configuration invalid!  Platform error sensor.hoymiles_dtu - No module named 'pymodbus.client.sync'

Mam nadzieję, że uda Ci się uaktualnić ten dodatek. Bardzo mi go brakuje.
Janusz

ArekKubacki, 2023-02-02:
Trwają pracę nad przeniesieniem na nowego pymodbusa. Aktualnie zalecam Downgrade do wersji styczniowej.

januszwoj, 2023-02-03:
Ok dzięki. Nie mogę się już doczekać nowszej wersji.

closed 2023-02-08 by ArekKubacki