ArekKubacki/Hoymiles-Plant-DTU-Pro #8: Exception: Invalid configuration for Hoymiles DTU platform

maickelr, 2023-02-08:
Hi! I have been working with your integration for a couple of months now, it has been working great. However since my DTU Pro got replaced by a new one (it stopped working) I have been having issues connecting. Home Assistant keeps throwing an error message: 

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 289, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 71, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

My configuration.yaml contains the following: 
 
sensor:
  - platform: hoymiles_dtu
    host: 192.168.1.60
    name: Hoymiles PV
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
    panels: 14

The DTU is connected to the LAN port. 
Any idea what could be wrong?

Many thanks for your efforts!!

ArekKubacki, 2023-02-13:
In this case, I have to guess. I bet the new DTU has some new firmware. My DTU works best with V00.02.10. For version 14 I couldn't even connect to modbus.

maickelr, 2023-02-14:
Thanks for your reply!  I had also read about issues with 2.14 but mine has firmware version V00.02.13 by the looks of it - so I did not really think that would be the problem.
But I understand it may be hard for you to troubleshoot like this.

No obvious things that I could check to find out where it goes wrong then :)

ArekKubacki, 2023-02-14:
Exactly :) However, I would downgrade to 10. They should send the appropriate files to the support.
Alternatively, the question whether the new DTU received the same IP address as the old one?

maickelr, 2023-02-28:
> 

Yes IP address is the same. I will contact support - see if they can downgrade. Thanks for your efforts on this!

closed 2023-02-28 by maickelr