ArekKubacki/Hoymiles-Plant-DTU-Pro #10: Pull interval

Matthijsbr123, 2023-04-05:
Hello,

Thnx for your great solution. I tried to pull the data in 1 minute, but it don't work. Any Tips or Tricks ?

sensor:
  - platform: hoymiles_dtu
    host: 192.168.X.X
    name: Hoymiles PV
    scan_interval: 00:01:00
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'

ArekKubacki, 2023-04-06:
Is the DTU connected via the ETH cable?

Matthijsbr123, 2023-04-06:
Yes

ArekKubacki, 2023-04-06:
That's one more question. Are you receiving data and want to increase the ic number every minute or are you not receiving data at all?

Matthijsbr123, 2023-04-06:
Yes i receive data, seems like every 5 minutes. I searched in the settings to speed up to 1 Minute

ArekKubacki, 2023-04-06:
Unfortunately they have changed it to HA for a while now. All sessions that are total type are refreshed every 5 minutes.

https://developers.home-assistant.io/docs/core/entity/sensor/

Matthijsbr123, 2023-04-08:
Better than 15 minutes. Thanks for your support !

ArekKubacki, 2023-04-13:
In the new version, everything except total and today production will no longer be in the measurement class, so you will be able to refresh more often.

closed 2023-04-28 by ArekKubacki