ArekKubacki/Hoymiles-Plant-DTU-Pro #15: Configuration.yaml

jrpfjr, 2023-07-11:
Hello. For the reading to work I must just include the following in the configuration.yaml 
  (sensor:
  - platform: hoymiles_dtu
    Host: 192.168.3.32
    name: Hoymiles PV
    dtu_type: 0
        monitored_conditions: [...])"

 or should I include some configuration of Home Assistant Modbus

(for example: Modbus:
  - name: "hub1"
    close_comm_on_error: true
    delay: 5
    Timeout: 5
        type: tcp) ?

ArekKubacki, 2023-07-12:
Hello,
platform: hoymiles_dtu is the answer

jrpfjr, 2023-07-12:
Hello.  Grateful.  By return.  I'm sorry, I'm new to Home Assistant.  What should I do?  Do I have to include the expression "modbus" and add-ons in configuration.yaml or just the expression "sensor" and add-ons in Hoymiles-Plant-DTU-Pro?

ArekKubacki, 2023-07-12:
under sensor.

ArekKubacki, 2023-07-12:
Does the DTU connect via cable? Have you activated modbus in the application?

ArekKubacki, 2023-07-14:
DTU must be connected by ETH not WiFi.

closed 2023-09-09 by ArekKubacki