ArekKubacki/Hoymiles-Plant-DTU-Pro #16: Error loading component in HA2023.9.0

hencou, 2023-09-07:
Since HomeAssistant 2023.9.0 this component is not working anymore and generates an error in the HA log: "cannot import name 'Defaults' from 'pymodbus.constants'".

As a temporary workaround I commented out lines 16-19 in this file https://github.com/ArekKubacki/Hoymiles-Plant-DTU-Pro/blob/main/custom_components/hoymiles_dtu/sensor.py, and now it seems working again. Can this lines safely being ignored or need it to be something else? 

Thank you in advance,

drac0linux, 2023-09-09:
I came here to report the same issue. I was hoping it would be fixed by the next update, however 2023.9.1 dropped a short while ago and the issue still exists afterwards. 

Will try the workaround by @hencou, would like to know the same if his workaround is problematic for the long term? I'll be happy to provide any log files if needed.

ArekKubacki, 2023-09-09:
The new version will be available on Monday at the latest.

ArekKubacki, 2023-09-09:
New Release on GitHub is available now!

closed 2023-09-09 by ArekKubacki