ArekKubacki/Hoymiles-Plant-DTU-Pro #17: Energy Spike after sunset

erwinvana, 2023-10-16:
Hello,
I have a spike after sunset, its not everyday , but its like 5 out of 7 days a week.
Apparently this is know issue with HA.
Form the Link:
[https://community.home-assistant.io/t/energy-dashboard-shows-crazy-and-unrealistic-energy-spike/479252/7](url)

**this can be resolved by:**
_integration developer should implement an “unavailable state” - or by creating a manual template sensor where you refer to the sensors state - but implement a check against zero-values._

i know i can correct this through the developer statistic option, but i don't think that is a long term solution.
Can you please advise in this ?

Screenshot of Issue:
[image]

ArekKubacki, 2023-10-16:
What variable are you using in Energy dashboard?

erwinvana, 2023-10-16:
[image]

ArekKubacki, 2023-10-16:
That is why such peaks occur. Change to **total** according to the information in the Readme and forget about the problem.

closed 2023-10-23 by ArekKubacki