ArekKubacki/Hoymiles-Plant-DTU-Pro #19: Entity is unavailable

hombre-13hb, 2023-11-01:
Hi,

on October 30th at around noon entity stopped working. I tried many restart but it keeps staying unavailable. 
Can you help ?
Thanks

ArekKubacki, 2023-11-01:
Send logo from HA.

MariuszSzurgot, 2023-11-11:
Hello same i have. Everyting work good and after HA update just stopped. below is the logs:

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:359
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 08:39:10 (1 occurrences)
Last logged: 08:39:10

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 76, in setup_platform
    dev.append(HoymilesPVSensor(name, updater.data.microinverter_data[i-1].serial_number, i, updater.data.microinverter_data[i-1].port_number, variable, updater))
                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
IndexError: list index out of range

ArekKubacki, 2023-11-11:
Have you restarted HA?

closed 2023-12-04 by ArekKubacki