ArekKubacki/Hoymiles-Plant-DTU-Pro #21: Home Assistant sensor sensor.yyy_hoymiles_pv_total_production dropped down by 400 kWh with DTU-Pro

FlySmack, 2023-12-11:
Hi,

some days ago (9th December) the Home Assistent sensor sensor.yyy_hoymiles_pv_total_production dropped down by 400 kWh.
Was at 1053,53 and went down to 616.21. Total Production sensor increases again during the day, but did not went back to the correct value.
If I remember well, the day I had installed the HASS Core update 2023.12.1.
S-Miles Cloud still reports the correct value. Restart of the DTU-Pro did not changed anything. The same for restarting the HASS instance.

DTU software version was V00.02.15. I will upgrade the DTU to V00.02.19 and check again.

Any idea about how to verify what goes wrong is highly welcome.

DTU-Pro (Gen 3) is collecting data from 2x HM-800 with 4 panels (2 panels for each HM-800).

ArekKubacki, 2023-12-11:
If HA is working, there is no possibility of the total value decreasing.  This is only possible with an HA restart.  If this happens, wait until the inverters wake up and report the new value.  Then you need to restart HA again.  Unfortunately, if the microinverters are sleeping and sent stupid messages in the last frame, there is nothing that can be done about it.

FlySmack, 2023-12-12:
Thanks for the advise. I will try what you have suggested and restart HA when both inverters are online (green LED flashing).

FlySmack, 2023-12-12:
I can confirm that this was the solution. All micro inverters must be online to ensure that you get proper values during startup /restart.
Thanks for the help.

closed 2023-12-12 by FlySmack