ArekKubacki/Hoymiles-Plant-DTU-Pro #23: Wrong value in Energy dashboard

erwinvana, 2024-01-25:
Since a couple of days, i have a strange value reported in teh HA Energy Dashboard.
I see 1 nagative spike in he morning (today -2,114.58 kWh), but no values the rest of the day
In the Hoymiles page in HA i can see teh output per panel and that looks fine.
[image]
[image]
[image]

ArekKubacki, 2024-01-27:
Have you restarted HA in the meantime?

erwinvana, 2024-01-27:
Hello ArekKubacki ,

Ive restarted HA multiple times, didnt solve it…

k.r.,
 Erwin

Verstuurd vanaf mijn iPhone

erwinvana, 2024-01-29:
I tested with the other vallue "Hoymiles PV today_production"  (which had other known issues in the end of the day) this seems to work. 
[image]

closed 2024-05-06 by ArekKubacki