ArekKubacki/Hoymiles-Plant-DTU-Pro #24: DTU Pro S -  Firmware 02.22

CrazymanXYZ, 2024-01-27:
Doesnt work anymore with DTU-Pro-S  Firmware 02.22.

Also Home Assistant triggers following info

`Logger: homeassistant.const
Source: helpers/deprecation.py:205
First occurred: 21:10:58 (12 occurrences)
Last logged: 21:10:58

DEVICE_CLASS_VOLTAGE was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.VOLTAGE instead, please report it to the author of the 'hoymiles_dtu' custom integration
DEVICE_CLASS_CURRENT was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.CURRENT instead, please report it to the author of the 'hoymiles_dtu' custom integration
TEMP_CELSIUS was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.CELSIUS instead, please report it to the author of the 'hoymiles_dtu' custom integration
DEVICE_CLASS_TEMPERATURE was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorDeviceClass.TEMPERATURE instead, please report it to the author of the 'hoymiles_dtu' custom integration
FREQUENCY_HERTZ was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfFrequency.HERTZ instead, please report it to the author of the 'hoymiles_dtu' custom integration`

CrazymanXYZ, 2024-01-27:
`Logger: homeassistant.components.sensor
Source: helpers/deprecation.py:205
Integration: Sensor (documentation, issues)
First occurred: 21:10:58 (6 occurrences)
Last logged: 21:10:58

STATE_CLASS_MEASUREMENT was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please report it to the author of the 'hoymiles_dtu' custom integration
STATE_CLASS_TOTAL was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL instead, please report it to the author of the 'hoymiles_dtu' custom integration
STATE_CLASS_TOTAL_INCREASING was used from hoymiles_dtu, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please report it to the author of the 'hoymiles_dtu' custom integration`

vriezenaar, 2024-01-28:
Getting the same.

ArekKubacki, 2024-01-28:
What you show will come into force in 2025.  There is no error here.  what exactly doesn't work.  If Hoymiles blocked Modbus, unfortunately you will have to downgrade or build OpenDTU for little money.  I use it myself and I don't remember the last time I had any problems.

CrazymanXYZ, 2024-01-29:
Got feedback from Hoymiles
Does this info would help you to publish an softwareupdate or a workaround for the affected people? :-) Really miss the readings from your software.

`GERMAN:
Die wichtigsten Modbus-Änderungen in der DTU-Firmware Version 22 sind:
1. Hinzufügen des Lesebefehls von Befehl 4 und der Datenpunkttabelle beginnend mit 0x3000 in der Modbus-Punktetabelle.
2. Hinzufügen der Tabelle mit den Informationen über Mikro-Wechselrichter-Alarme in der Modbus-Punktetabelle. 

ENGLISH:
The main Modbus changes in DTU firmware version 22 are:
1. Add the read command of command 4 and the data point table starting from 0x3000 in the Modbus point table.
2. Adding the microinverter alarm information table in the Modbus points table.
`

CrazymanXYZ, 2024-01-29:
At the moment no readings can be read 
[image]
[image]

ArekKubacki, 2024-01-29:
Can you edit files on your HA? I would give you the line number where you could check its operation.

CrazymanXYZ, 2024-01-29:
I am a noobie, but can do my best and give it a try :-)

ArekKubacki, 2024-01-29:
homeassistant/custom_components/hoymiles_dtu/hoymiles/client.py
Change in line 114 and 116 
0x1000 to 0x3000

biasowicka, 2024-02-01:
podpinam temat

alarta, 2024-02-02:
Hi ArekKubacki
you have any more ideas how to fix?

ArekKubacki, 2024-02-07:
Please send the logs. Mostly errors, not warnings.

alarta, 2024-02-07:
with client.py:114/116
start_address = i * 20 + 0x1000

Logger: pymodbus.logging
Source: custom_components/hoymiles_dtu/hoymiles/client.py:98
Integration: hoymiles_dtu (documentation)
First occurred: 13:32:19 (2 occurrences)
Last logged: 13:32:24
Connection to (192.168.1.205, 502) failed: timed out

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:360
Integration: Sensor (documentation, issues)
First occurred: 13:32:24 (1 occurrences)
Last logged: 13:32:24

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

with client.py:114/116
start_address = i * 20 + 0x3000

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:408
Integration: Sensor (documentation, issues)
First occurred: 13:38:00 (1 occurrences)
Last logged: 13:38:00

Setup of platform hoymiles_dtu is taking longer than 60 seconds. Startup will proceed without waiting any longer.

Logger: homeassistant
Source: custom_components/hoymiles_dtu/sensor.py:69
Integration: hoymiles_dtu (documentation)
First occurred: 13:38:05 (1 occurrences)
Last logged: 13:38:05

Error doing job: Future exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

ArekKubacki, 2024-02-11:
I am currently waiting for a response from Hoymiles technicians. Unfortunately, it is currently Chinese New Year.

biasowicka, 2024-02-12:
I asked Hoymiles support to downgrade DTU PRO-S to version 2.19, and it started working

ArekKubacki, 2024-02-12:
Can you attach files for downgrade? Personally, I have an old version in my
files.

biasowicka, 2024-02-12:
They did the downgrade remotely because I asked them for it, I also asked for a share of the software, as soon as they send me the file, I will post it here.

biasowicka, 2024-02-12:
Firmware DTU PRO-S to version 2.19 in the attachment.

[Uploading V0.2.19 DTU Pro or Pro S.zip…]()

biasowicka, 2024-02-12:
Firmware 
[V0.2.19 DTU Pro or Pro S.zip](https://github.com/ArekKubacki/Hoymiles-Plant-DTU-Pro/files/14237645/V0.2.19.DTU.Pro.or.Pro.S.zip)
DTU PRO-S to version 2.19

alarta, 2024-02-20:
from photovoltaikforum:
But I now know what the problem is: In the old version, the registers can be read with "Read Input Register" (CMD 4) and with "Read Holding Register" (CMD 3). In V2.22 only “Read Holding Register (CMD 3) works”.

alarta, 2024-02-21:
got it working ... bug from Hoymiles
For 22 Version, if you choose "Export control", it prohibits MODBUS TCP. But "Remote control" still supports it.
So you have to go to "Toolkit" in APP and then choose "Remote Control" from RS-485 Port Config under DTU Information.

ArekKubacki, 2024-02-22:
Thank you for finding a workaround. This solution actually works. I took the liberty of using your solution in the Readme file in the repository.

closed 2024-02-22 by ArekKubacki

bynight66, 2024-05-02:
> got it working ... bug from Hoymiles For 22 Version, if you choose "Export control", it prohibits MODBUS TCP. But "Remote control" still supports it. So you have to go to "Toolkit" in APP and then choose "Remote Control" from RS-485 Port Config under DTU Information.

What is the correct Portaddress?

vriezenaar, 2024-05-03:
502 is the port that is used (coded in the software). I have also .19
[image]
Mine works with "export control". Connected via Ethernet. Are you sure you have the right IP? Check the mac adres in your router to check.

bynight66, 2024-05-06:
I have the .22 now and it works with remote Control. But there I have to choose a Portaddress... from 101 ~ 254...
Export Control I did not try...
[image]

alarta, 2024-05-06:
i used 101