ArekKubacki/Hoymiles-Plant-DTU-Pro #25: Hoymiles_modbus 0.6.0 error with HA 2024.2: Python version incompatible

FvanEgmond, 2024-02-08:
HA 2024.2 ships with Python 3.12; 
Hoymiles_modbus version 0.6.0 only works with Python 3.8-3.11.
Therefore the following error is given:  

_Unable to install package hoymiles_modbus>=0.6.0: 
ERROR: Ignored the following yanked versions: 0.3.0 
ERROR: Ignored the following versions that require a different python version: 0.6.0 
Requires-Python >=3.8,<3.12; 0.6.1 
Requires-Python >=3.8,<3.12; 0.6.2 
Requires-Python >=3.8,<3.12 
ERROR: Could not find a version that satisfies the requirement hoymiles_modbus>=0.6.0 (from versions: 0.1.0, 0.2.0, 0.2.1, 0.4.0, 0.5.0) 
ERROR: No matching distribution found for hoymiles_modbus>=0.6.0_

biasowicka, 2024-02-10:
Logger: homeassistant.util.package
Source: util/package.py:122
First occurred: 9 lutego 2024 19:18:28 (6 occurrences)
Last logged: 12:31:55

Unable to install package hoymiles_modbus>=0.6.0: ERROR: Ignored the following yanked versions: 0.3.0 ERROR: Ignored the following versions that require a different python version: 0.6.0 Requires-Python >=3.8,<3.12; 0.6.1 Requires-Python >=3.8,<3.12; 0.6.2 Requires-Python >=3.8,<3.12 ERROR: Could not find a version that satisfies the requirement hoymiles_modbus>=0.6.0 (from versions: 0.1.0, 0.2.0, 0.2.1, 0.4.0, 0.5.0) ERROR: No matching distribution found for hoymiles_modbus>=0.6.0

hpsteff, 2024-02-10:
same here: 
would love to be able to use the tool to monitor my solar installation  further.
---
Logger: homeassistant.util.package
Source: util/package.py:122 
First occurred: 12:08:32 (3 occurrences) 
Last logged: 12:09:08

Unable to install package hoymiles_modbus>=0.6.0: ERROR: Ignored the following yanked versions: 0.3.0 ERROR: Ignored the following versions that require a different python version: 0.6.0 Requires-Python >=3.8,<3.12; 0.6.1 Requires-Python >=3.8,<3.12; 0.6.2 Requires-Python >=3.8,<3.12 ERROR: Could not find a version that satisfies the requirement hoymiles_modbus>=0.6.0 (from versions: 0.1.0, 0.2.0, 0.2.1, 0.4.0, 0.5.0) ERROR: No matching distribution found for hoymiles_modbus>=0.6.0

thanks for looking into this.
Hans-Peter

ArekKubacki, 2024-02-10:
Check new release.

biasowicka, 2024-02-10:
nie omieszkam zainstalować i jutro przy słońcu dam znać

biasowicka, 2024-02-11:
Po aktualizacji nie ma juz problemów z wersja Pyth, ale niestety po restarcie nie dziala :

Logger: homeassistant
Source: custom_components/hoymiles_dtu/sensor.py:69
Integration: hoymiles_dtu ([documentation](https://github.com/ArekKubacki/Hoymiles-Plant-DTU-Pro))
First occurred: 11:55:29 (1 occurrences)
Last logged: 11:55:29

Error doing job: Future exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

Logger: homeassistant.bootstrap
Source: bootstrap.py:566
First occurred: 11:55:05 (1 occurrences)
Last logged: 11:55:05

Waiting on integrations to complete setup: hoymiles_dtu.sensor

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:392
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 11:55:21 (1 occurrences)
Last logged: 11:55:21

Setup of platform hoymiles_dtu is taking longer than 60 seconds. Startup will proceed without waiting any longer.

ArekKubacki, 2024-02-11:
Why should it work if this release, as described, only fixes the python version problem, and in another topic you wrote that you updated DTU to the latest version? Please use English in English threads. The thread in Polish is in another place.

The new release fixes the problem described in this thread.

closed 2024-02-11 by ArekKubacki

FvanEgmond, 2024-02-11:
I can confirm that the new version solves this issue. :-) Thank you for the quick fix. Cheers!