ArekKubacki/Hoymiles-Plant-DTU-Pro #27: Warnings on Depricated constants

FvanEgmond, 2024-03-07:
Home Assistant 2024 versions are giving warnings on deprecated constants (see list below). 
I know this is not a breaking change -and it will only stop working in 2025.1- but for the completeness and correctness, this issue is created, so it can be tracked and addressed somewhere this year. :-)
Thanks in advance for fixing! 

- STATE_CLASS_MEASUREMENT -> Use SensorStateClass.MEASUREMENT instead
- STATE_CLASS_TOTAL -> Use SensorStateClass.TOTAL instead
- STATE_CLASS_TOTAL_INCREASING -> Use SensorStateClass.TOTAL_INCREASING instead
- DEVICE_CLASS_VOLTAGE -> Use SensorDeviceClass.VOLTAGE instead
- DEVICE_CLASS_CURRENT -> Use SensorDeviceClass.CURRENT instead
- TEMP_CELSIUS -> Use UnitOfTemperature.CELSIUS instead
- DEVICE_CLASS_TEMPERATURE -> Use SensorDeviceClass.TEMPERATURE instead
- FREQUENCY_HERTZ -> Use UnitOfFrequency.HERTZ instead

ArekKubacki, 2024-04-10:
Fixed in new release.

closed 2024-04-10 by ArekKubacki

FvanEgmond, 2024-04-15:
Thank you so much. It works perfectly. :-)