ArekKubacki/Hoymiles-Plant-DTU-Pro #28: looks really good... but some questions...

bynight66, 2024-05-02:
I spend a lot of time with a not working 2 years old SW from a banny... :-(

1. Does the Hoymiles SW still work beside this SW from you? Or do I have to choice between HM SW and your SW?
2. It's possible to change the output power rate (%)? It's important for using Hoymiles on Victrons ACout...
3. Do you still work on this SW?
4. I have 30-40 HMs... The graphic looks good... Is this possible with HA only, without Grafana or so?

ArekKubacki, 2024-05-06:
1. I don't know because I don't use both.
2. Theoretically yes, but it would involve a huge configuration because you would have to specify the maximum power for each panel.
3. Yes, I work.
4. I put a sample graphic in the readme. You have to customize its appearance yourself. You can also skip it and work on entities.

bynight66, 2024-05-06:
1. Yes it  works. I like and use the DTU Pro and all inverters are visible and I get all Date in HA. -> very good!
2. I don't think that it needs a huge configuration for each Panel. I think there is only one value, which needs to be changed, but this is a very impotant function: 
[image]
4. Thanks a lot for this sample sample grafik! I modified it already and it works.  
[image]
Not so nice as the original and I dont know if I need this in future... because the Plant Power and Energy I get from an Shelly 3EM after the Hoymiles very easy and fast. And the 3 EM communicate with Victron Cerbo and with HA.

But what is always missing and what nobody has is the Power Regulation!