ArekKubacki/Hoymiles-Plant-DTU-Pro #30: Problem z konfiguracją

Mazurekmazi, 2024-05-06:
Witam.

Problem z konfiguracją DTU Pro s połączenie eth, konfiguracja jak w instrukcji
  sensor:
    - platform: hoymiles_dtu
      host: 192.168.1.234
      name: Hoymiles PV
      dtu_type: 0
      monitored_conditions:
        - 'pv_power'
        - 'today_production'
        - 'total_production'
        - 'alarm_flag'
      monitored_conditions_pv:
        - 'pv_power'
        - 'today_production'
        - 'total_production'
        - 'pv_voltage'
        - 'pv_current'
        - 'grid_voltage'
        - 'temperature'
        - 'operating_status'
        - 'alarm_code'
        - 'alarm_count'
        - 'link_status'
      panels: 6
 Błąd jak poniżej.

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 356, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

Pozdrawiam

ArekKubacki, 2024-05-06:
A topic in Polish has been created. In the rest, the standard language is English.

Are you sure IP is eth? Sometimes dtu connects both via wifi and eth. You need to be sure.

Have you followed?:
!!! For 22 Version, if you choose "Export control", it prohibits MODBUS TCP. But "Remote control" still supports it. So you have to go to "Toolkit" in APP and then choose "Remote Control" from RS-485 Port Config under DTU Information. !!!

Mazurekmazi, 2024-05-06:
I haven't configured the WiFi, so it's probably connected via eth, the IP address is set on the Mikrotik, so the ports are probably entered correctly, and the ones that need to be redirected, I haven't set up export management, and the error is still displayed.

Mazurekmazi, 2024-05-06:
I fix the problem. Thx

closed 2024-05-06 by Mazurekmazi