ArekKubacki/Hoymiles-Plant-DTU-Pro #33: Integration with my H09.02.02 DTU

gmatrangola, 2024-10-23:
### The Hoymiles sensor entities do not appear in my HomeAssistant instance.

This is what I did to install the Integration:
1. I added the custom repository in hacs and added your URL.
2. I  used HACS to download the integration
3. The Integration did not appear in `Settings->Devices & Services -> Integrations`
4.  So I clicked add. It was not available for add.
5. I rebooted my Home Assistant
6. `Settings->Devices & Services -> Integrations` was able to click Click on `Hoymiles Plant DTU-PRO` in `Add Integration`
7. When I try to add Add Integration I get an error dialog: "This device cannot be added from the UI You can add this device by adding it to your 'configuration.yaml'. See the documentation for more information."
8. So I confirmed that the `custom_components/hoymiles_dtu` folder was on my filesystem
9. Then I copied the Example Configuration from the Github into my `configuraiton.yaml` file using **Studio Code Server**
10. I updated the _host IP_ address, panels count and saved the yaml file.
11. I rebooted my Home Assistant
12. I created a new **Solar Dashboard**
13. I added an **Entity Card Configuration** card to the Dashboard
14. But when I search for _hoymiles_ as the entity, the only entity that appears is the **Update** entity.

marcioaneves, 2025-01-08:
Hi, 
I have the same situation here. For my understanding,  Do you have a example about the step 9. Where Do I need to add in configuration.yaml ?

ArekKubacki, 2025-01-09:
Have you ever edited the configuration.yaml file? Do you know how to get to it?

If this is your first time with the file, look for a tutorial on the Internet.

marcioaneves, 2025-01-09:
I can edit the configuration file, but I need guidance on where to add this within configuration.yaml. Could you point me to the correct section?  
I added it but the integration was not enable. 

sensor:
  - platform: hoymiles_dtu
    host: 192.168.15.10
    name: Hoymiles PV
    dtu_type: 0
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: 16

ArekKubacki, 2025-01-09:
Looks correct. Make sure you don't use the word sensor twice.

marcioaneves, 2025-01-09:
There is another section as find below, but is is in the mqtt. 

# # OS Sensor
mqtt:
  sensor:
     - name: OS_stats
      state_topic: "hiveos"
      value_template: "{{ value_json.params.miner_stats.uptime }}"
      json_attributes_topic: "hiveos"
      json_attributes_template: "{{ value_json.params.miner_stats | to_json }}"

hugovdkort, 2025-02-07:
I too struggle with getting your integration running. Can't find good guides online with this particular problem. I'm a total noob with home assistant, but I unfortunately am stuck with hoymiles....

See my yaml below. I should be able to find the sensors in the entities browser even if there is no connection, right?

[image]

closed 2025-02-07 by ArekKubacki