ArekKubacki/Hoymiles-Plant-DTU-Pro #35: Getting S/N and Port #

ccolumbu999, 2024-12-07:
Hi  ArekKubacki,
I have a HoyMiles install and I have been playing with the ModBus TCP directly.
I figured out how to get the S/N and Port numbers from the data.
Here is how I did it.
You need values 0-3, convert each of them to binary. Zero pad so they are length 16.
So now you will have 4 values that look like this: 0000110000010001
Concatenate them together so you have something like this:
0000110000010001000011000001000100001100000100010000110000010001
Next split them into chunks of 4 so you have:
[0] = 0000
[1] = 1100
[2] = 0001
[3] = 0001
. . . 
[15] = 0001
Convert chunks 2-13 to decimal and you will have the S/N, convert chunk 15 to decimal and you will have the port #.
Please let me know if you have any questions, I hope this helps so you can include S/N + Port in a future version.

ArekKubacki, 2024-12-09:
Hello,
Please tell me why you need the S/N information in HA? Currently I don't see any point in creating an entity in HA that doesn't change and has a constant S/N number.

ccolumbu999, 2024-12-09:
The idea is to show the production of each panel and their performance over time. You show each panel, but don't identify it by S/N + Port which is what you would need if a panel needs service or replacing.
I don't use this for myself at all, I was just sharing a way for you to get the info to improve other user's experience. I am writing my own custom tracking system unrelated to HA.

ArekKubacki, 2024-12-09:
The entity name for each panel is sensor.hoymiles_pv_'sn'_pv_'port'_pv_power. Each panel has exact identification.

ccolumbu999, 2024-12-09:
Ok cool

closed 2024-12-09 by ccolumbu999