ArekKubacki/Hoymiles-Plant-DTU-Pro #36: Error since HA 1.2025

peterophet, 2025-01-04:
Platform error 'sensor' from integration 'hoymiles_dtu' - No module named 'pymodbus.framer.socket_framer'

I have this error since I updated to HA 1.2025. More people having this problem?

ArekKubacki, 2025-01-04:
Work in progress...

peterophet, 2025-01-04:
Thank you! Still a satisfied user of your integration. Works well with the Hoymiles DTU.

marisvs, 2025-01-05:
I have the same error. It stopped working after HA Core update to 2025.1.0. Had to restore the previous version of HA Core since  I don't want to miss the Solar energy information. I saw in the changelog of HA Core 2025.1.0 the following message "changed pymodbus module (Bump pymodbus version 3.7.4)".
See https://github.com/home-assistant/core/pull/133175

ArekKubacki, 2025-01-05:
After all, a corrected version has been available since yesterday. Version
0.7.0.

peterophet, 2025-01-05:
That is great and fast.
Go to HACS - click on the Hoymiles DTU - download again - download - restart Home Assistant
Up and running again

kneschi2, 2025-01-06:
For me, the plugin no longer starts after the update. The entity is no longer available. When I restart, the message about starting the DTU_Hoymiles plugin no longer appears.

marisvs, 2025-01-07:
Thanks for the fix 0.7.0. I have been able to increase to HA Core 2025.1.0 and apply 0.7.0 and I have now again the Solar panel information as before.

ArekKubacki, 2025-01-07:
> Still not working for me:

> For me, the plugin no longer starts after the update. The entity is no longer available. When I restart, the message about starting the DTU_Hoymiles plugin no longer appears.

Please provide the most important information, i.e. HA version and version of the DTU module installed in HACS.

Golempl, 2025-01-07:
If you need anything more, let me know.
My config:
Core 2025.1.1
Supervisor 2024.12.3
Operating System 14.1
Interfejs użytkownika 20250106.0

Hoymiles Plant DTU-Pro 0.7.0

Rejestrator: homeassistant.components.sensor
Źródło: helpers/entity_platform.py:366
integracja: Sensor (dokumentacja, Problemy)
Pierwsze zdarzenie: 11:10:46 (1 zdarzenia)
Ostatnio zalogowany: 11:10:46
Error while setting up hoymiles_dtu platform for sensor

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

sensor:
  - platform: hoymiles_dtu
    host: 192.168.1.149
    name: Hoymiles PV
    dtu_type: 0
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: 12

ArekKubacki, 2025-01-07:
This means that the module is working but there is no communication with the DTU. Did it work correctly before? Did you reset the HA several times? Sometimes the DTU does not connect to the 1st time.

Golempl, 2025-01-07:
It's been working for months without problem, just after last update of HA, i have rebooted DTU , checked if it's reachable and seems ok , but still unavailable

winbard, 2025-01-08:
Hello,

first thank for you very good work here.
Than I wish a happy new year.

I have the same problem that i get no more information in Homeassitant since the new firmware fronm the DTU.
When I look in settings-hacs-devices I see only this

[image]

When you need more information please write me.

Greets Klaus

ArekKubacki, 2025-01-09:
To all people reporting the bug. Please provide additionally:
HA version:
DTU firmware version:
Hoymiles custom component version:

peterophet, 2025-01-09:
https://support.hoymiles.com/hc/en-001/articles/11661871091983-S-Miles-Cloud-Update-Notice
It could be the new DTU firmware
I use 4.03 and it works fine.

kneschi2, 2025-01-09:
I have DTU-Pro V00.02.19 and with HA 2024.12.5, HACS 2.0.2 , DTU-Plugin v0.6.2  is working ok.
Whit HA 2025.1.1 and DTU-Plugin v0.7.0 the plugin does not start.
If I restore a backup the plugin works again
This means that the error should not be due to the DTU-Pro

alarta, 2025-01-09:
Hi
Having same issue.
DTU is online, responds to ping.
Tested with separate modbus tool - modbus working and receiving data.
Tried changing DTU Pro firmware 02.19 to 02.23 to 05.02

HA version: 2025.1.1
DTU firmware version: V00.05.02
Hoymiles custom component version: 0.7.0

Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

ArekKubacki, 2025-01-09:
I just checked both OpenDTU and my DTU and both work. Did you change the settings as on the main page?

!!! For 22 Version, if you choose "Export control", it prohibits MODBUS TCP. But "Remote control" still supports it. So you have to go to "Toolkit" in APP and then choose "Remote Control" from RS-485 Port Config under DTU Information. !!!

alarta, 2025-01-09:
yeah, "funny" thing is that i can't access my DTU through APP, it just connects to DTUs wifi and refuses to connect.
But nothing else but HAs ver changed before it stopped working. Didn't update or change anything in DTU.
I have OpenDTU as well and I know Hoymiles DTUPro sucks, but i really like its statistics and overview.
Shame you can't use both simultaneously.

winbard, 2025-01-09:
Hello,

H09.06.11
V00.05.02
DTU Pro-S

Where can I find the Hoymiles custom component version?

I try to install it manual but not work too.

alarta, 2025-01-09:
Managed to get in DTU and "Remote Control" is still checked.
Any more ideas?
Connect directly from HA to modbus?

morrowwm, 2025-01-09:
Version 0.7.0 working fine here. Much more reliable than some other integrations using wi-fi.
HA version: 2025.1.1
DTU firmware version: V00.02.23
Hoymiles custom component version: 0.7.0

Jarmo06, 2025-01-10:
Hello,
This (below) setup works for me, but when I updated to version 2025.1.1 / Plugin DTU-Pro v0.7.0, it stopped working, so I reverted back to the older 2024.12.5 / Plugin DTU-Pro v0.6.2 version.

A working system:

HA Core  2024.12.5
HA Supervisor 2024.12.3
DTU-Pro-S
DTU firmware version: V00.02.23
DTU Hardware version: H09.06.01
Plugin DTU-Pro v0.6.2

winbard, 2025-01-10:
Hello,

the entities are not provided.

[image]

xmatr1x69, 2025-01-10:
> To all people reporting the bug. Please provide additionally: HA version: DTU firmware version: Hoymiles custom component version:

HA version:
      Core: 2025.1.2
      Supervisor: 2024.12.3
      Operating System: 14.1
DTU firmware version: V15.04.09
DTU Hardware version: H09.01.01
Hoymiles custom component version: 0.7.0

I have remote control, before update it was working without a problem.
sensors.yaml
`- platform: hoymiles_dtu
  host: 192.168.1.5
  name: Hoymiles PV
  dtu_type: 0
  monitored_conditions:
    - 'pv_power'
    - 'today_production'
    - 'alarm_flag'
  monitored_conditions_pv:
    - 'pv_power'
    - 'grid_voltage'
    - 'temperature'
    - 'operating_status'
    - 'alarm_code'
  panels: 20
  scan_interval: "00:02:00"`
  
  I added it into configuration.yaml too if !include could be a problem:
  `sensor:
  - platform: hoymiles_dtu
    host: 192.168.1.5
    name: Hoymiles PV
    dtu_type: 0
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
    panels: 20
    scan_interval: "00:02:00"`
    
I can see entities, but they are not updating. Hoymiles app shows 135W power (so its not 0 all the time)
Nothing in the logs...
`2025-01-10 14:19:30.836 INFO (MainThread) [homeassistant.components.sensor] Setting up hoymiles_dtu.sensor
2025-01-10 14:19:30.836 INFO (MainThread) [homeassistant.setup] Setup of domain sensor took 0.00 seconds
2025-01-10 14:29:46.658 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.hoymiles_pv_pv_power is taking over 10 seconds`

I can ping hoymiles dtu pro.

Pymodbus is working.

One thing with hoymiles in logs:
`2025-01-10 14:36:13.964 DEBUG (MainThread) [homeassistant.bootstrap] Integration remaining: defaultdict(<class 'float'>, {'hoymiles_dtu': 7.28199107083492, 'hon': 6.56720941583626})`

Idk what to do. Tell me if you need any more logs/info. Was working great before.

winbard, 2025-01-12:
Hello,

I have try some.
I delete all from your program.
Entities too.
Than I install new.
After new install i see no entities under settings devices.
I have only the update entitie.

wouterDevelop, 2025-01-12:
same problem over here.  After I update Homeassistant to 2025.1.2 and Update the integration to 0.7.0, several reboots not working 

Just update Ha to version 

Core: 2025.1.2
Supervisor: 2024.12.3
Operating System: 14.1
Frontend: 20250109.0 

intergration : 0.7.0

config: 

  - platform: hoymiles_dtu
    host: 192.168.111.10
    name: Hoymiles PV
    dtu_type: 0
    panels: 12
    monitored_conditions:
      - "pv_power"
      - "today_production"
      - "total_production"
      - "alarm_flag"
    monitored_conditions_pv:
      - "pv_power"
      - "today_production"
      - "total_production"
      - "pv_voltage"
      - "pv_current"
      - "grid_voltage"
      - "temperature"
      - "operating_status"
      - "alarm_code"
      - "alarm_count"
      - "link_status"
      
   Error: 

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

winbard, 2025-01-13:
Hello,

please excuse me.
I read wrong first.
My HA Version is 2.205
Since 2025 vision no values are could read from Hoymiles.
I make backup to 2024 and it work.
I had another problem with DTU Firmare 00.05.02. In the app and web from Hoymiles some microconverters will not read back. The Layout work but not for. ex. the power read from the microconverter. But this is Hoymiles problem. With the Version before it work.
But that had nothing to do with your program. It only work with HA 2024 versions. Not with 2025 versions

alarta, 2025-01-13:
> I Increased the time it takes to read data from DTU. I Increased the number of read attempts. Check out version 0.7.1.

still same error for me too
checked and all variables are correct in line 66
any more possibilities to debug?

ArekKubacki, 2025-01-13:
This is very strange. For many people, including me, everything works. Here, it is clearly not a bug in the code, but a lack of information downloaded from DTU. But why does it work in the previous version? There is an older modbus library there and it works a bit differently.

People who have access to their computers outside HA, please upload and run the test program in python. Remember to change the IP address in the my_script.py file.
After unpacking, run the command python3 my_script.py
[hoymiles_test.zip](https://github.com/user-attachments/files/18397063/hoymiles_test.zip)

Golempl, 2025-01-13:
Tested on:
 Core 2025.1.2
Supervisor 2024.12.3
Operating System 14.1
Interfejs użytkownika 20250109.0

Hoymiles Plant DTU-Pro
version": "0.7.1"

configuration.yaml
sensor:
  - platform: hoymiles_dtu
    host: 192.168.1.149
    name: Hoymiles PV
    dtu_type: 0
    monitored_conditions:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'alarm_flag'
    monitored_conditions_pv:
      - 'pv_power'
      - 'today_production'
      - 'total_production'
      - 'pv_voltage'
      - 'pv_current'
      - 'grid_voltage'
      - 'temperature'
      - 'operating_status'
      - 'alarm_code'
      - 'alarm_count'
      - 'link_status'
    panels: 16

Problem still exist.

2025-01-13 13:21:15.090 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up hoymiles_dtu platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

While addon start it connects to DTU, and DTU gives this data to HomeAssistant:
(Packet is received,HA confirms tcp ack accepting packets from DTU)

Modbus/TCP
    Transaction Identifier: 1
    Protocol Identifier: 0
    Length: 43
    Unit Identifier: 1
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    [Request Frame: 7]
    [Time from request: 0.012007000 seconds]
    Byte Count: 40
    Register 4096 (UINT16): 3091
        Register Number: 4096
        Register Value (UINT16): 3091
    Register 4097 (UINT16): 33424
        Register Number: 4097
        Register Value (UINT16): 33424
    Register 4098 (UINT16): 24592
        Register Number: 4098
        Register Value (UINT16): 24592
    Register 4099 (UINT16): 26625
        Register Number: 4099
        Register Value (UINT16): 26625
    Register 4100 (UINT16): 310
        Register Number: 4100
        Register Value (UINT16): 310
    Register 4101 (UINT16): 0
        Register Number: 4101
        Register Value (UINT16): 0
    Register 4102 (UINT16): 2293
        Register Number: 4102
        Register Value (UINT16): 2293
    Register 4103 (UINT16): 4998
        Register Number: 4103
        Register Value (UINT16): 4998
    Register 4104 (UINT16): 2
        Register Number: 4104
        Register Value (UINT16): 2
    Register 4105 (UINT16): 1
        Register Number: 4105
        Register Value (UINT16): 1
    Register 4106 (UINT16): 5
        Register Number: 4106
        Register Value (UINT16): 5
    Register 4107 (UINT16): 12237
        Register Number: 4107
        Register Value (UINT16): 12237
    Register 4108 (UINT16): 31
        Register Number: 4108
        Register Value (UINT16): 31
    Register 4109 (UINT16): 3
        Register Number: 4109
        Register Value (UINT16): 3
    Register 4110 (UINT16): 9
        Register Number: 4110
        Register Value (UINT16): 9
    Register 4111 (UINT16): 0
        Register Number: 4111
        Register Value (UINT16): 0
    Register 4112 (UINT16): 263
        Register Number: 4112
        Register Value (UINT16): 263
    Register 4113 (UINT16): 0
        Register Number: 4113
        Register Value (UINT16): 0
    Register 4114 (UINT16): 0
        Register Number: 4114
        Register Value (UINT16): 0
    Register 4115 (UINT16): 0
        Register Number: 4115
        Register Value (UINT16): 0

Golempl, 2025-01-13:
hoymiles_test python3 my_script.py 
Traceback (most recent call last):
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/my_script.py", line 2, in <module>
    from hoymiles.client import HoymilesModbusTCP
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/hoymiles/client.py", line 5, in <module>
    from ._modbus_tcp_client import create_modbus_tcp_client
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/hoymiles/_modbus_tcp_client.py", line 5, in <module>
    from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.pdu.register_read_message'

Added pymodbus using pymodbus because was missing, but still can't run this test tool

(venv) ➜  hoymiles_test pip install --upgrade pip
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Requirement already satisfied: pip in ./venv/lib/python3.12/site-packages (24.3.1)
(venv) ➜  hoymiles_test pip install pymodbus
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Requirement already satisfied: pymodbus in ./venv/lib/python3.12/site-packages (3.8.3)
(venv) ➜  hoymiles_test

ArekKubacki, 2025-01-13:
pip3 install pymodbus==3.7.4
and one more time :)

Golempl, 2025-01-13:
ok ,
hoymiles_test pip show pymodbus           
Name: pymodbus
Version: 3.7.4
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: Galen Collins, Jan Iversen
Author-email: 
License: BSD-3-Clause
Location: /homeassistant/custom_components/hoymiles_dtu/hoymiles_test/venv/lib/python3.12/site-packages
Requires: 
Required-by: 
(venv) ➜  hoymiles_test pip show plum-py 
Name: plum-py
Version: 0.8.7
Summary: Pack/Unpack Memory.
Home-page: https://plum-py.readthedocs.io/en/latest/index.html
Author: Dan Gass
Author-email: dan.gass@gmail.com
License: MIT License; http://opensource.org/licenses/MIT
Location: /homeassistant/custom_components/hoymiles_dtu/hoymiles_test/venv/lib/python3.12/site-packages
Requires: 
Required-by:

While running:

python3 my_script.py 

Traceback (most recent call last):
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/venv/lib/python3.12/site-packages/plum/data.py", line 209, in unpack
    item, offset = cls.__unpack__(buffer, 0, None)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 236, in __unpack__
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/hoymiles/datatypes.py", line 78, in _pv_current_type
    raise ValueError(f"Couldn't detect inverter type for serial {serial}. Please report an issue.")
ValueError: Couldn't detect inverter type for serial 138290601068. Please report an issue.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/venv/lib/python3.12/site-packages/plum/data.py", line 228, in unpack_and_dump
    item, offset = cls.__unpack__(buffer, 0, dmp.add_record(fmt=cls.name))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 274, in __unpack__
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/hoymiles/datatypes.py", line 78, in _pv_current_type
    raise ValueError(f"Couldn't detect inverter type for serial {serial}. Please report an issue.")
ValueError: Couldn't detect inverter type for serial 138290601068. Please report an issue.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/my_script.py", line 5, in <module>
    plant_data = HoymilesModbusTCP('192.168.1.149', port=502, unit_id=modbusID).plant_data
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/hoymiles/client.py", line 104, in plant_data
    inverters = self.inverters
                ^^^^^^^^^^^^^^
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/hoymiles/client.py", line 78, in inverters
    inverter_data = InverterData.unpack(data_to_unpack)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/venv/lib/python3.12/site-packages/plum/data.py", line 216, in unpack
    cls.unpack_and_dump(buffer)
  File "/homeassistant/custom_components/hoymiles_dtu/hoymiles_test/venv/lib/python3.12/site-packages/plum/data.py", line 242, in unpack_and_dump
    raise exceptions.UnpackError(dmp, exc) from exc
plum.exceptions.UnpackError: 

+--------+---------------+-----------------------+-------------------+--------------------------+
| Offset | Access        | Value                 | Bytes             | Format                   |
+--------+---------------+-----------------------+-------------------+--------------------------+
|        |               |                       |                   | InverterData (Structure) |
|  0     | data_type     | 12                    | 0c                | uint8                    |
|        | serial_number |                       |                   | serial_number_t          |
|  1     |   [0:6]       | b'\x13\x82\x90`\x10h' | 13 82 90 60 10 68 |                          |
|  7     | port_number   | 1                     | 01                | uint8                    |
|  8     | pv_voltage    | 0                     | 00 00             | udec16p1                 |
+--------+---------------+-----------------------+-------------------+--------------------------+

ValueError occurred during unpack operation:

Couldn't detect inverter type for serial 138290601068. Please report an issue.

Golempl, 2025-01-13:
I have updated DTU firmware but it was days after problem existed.

DTU	SN4121A001DE16	Hardware Ver.H09.06.11 Firmware Ver.V00.05.02
And (there was no updates:
Micro	SN138290601007	Model HMT-2250-6T Hardware Ver.H00.04.00	Firmware Ver.V01.00.10
Micro	SN138290601068	Model HMT-2250-6T Hardware Ver.H00.04.00	Firmware Ver.V01.00.10
And seems theres problem with processing Micro

Golempl, 2025-01-13:
[image]

Golempl, 2025-01-13:
Added:
        current_type = _PVCurrentType.HM.value
    elif serial.startswith('13'):
    
    at datatypes.py
    and now:
    hoymiles_test python3 my_script.py
PlantData(dtu='4121a001de16', pv_power=Decimal('0'), today_production=0, total_production=0, alarm_flag=False, inverters=[InverterData(data_type=12, serial_number='138290601068', port_number=1, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601068', port_number=2, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601068', port_number=3, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601068', port_number=4, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601068', port_number=5, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601068', port_number=6, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601007', port_number=1, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601007', port_number=2, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601007', port_number=3, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601007', port_number=4, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601007', port_number=5, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0]), InverterData(data_type=12, serial_number='138290601007', port_number=6, pv_voltage=Decimal('0'), pv_current=Decimal('0'), grid_voltage=Decimal('0'), grid_frequency=Decimal('0'), pv_power=Decimal('0'), today_production=0, total_production=0, temperature=Decimal('0'), operating_status=0, alarm_code=9, alarm_count=0, link_status=0, reserved=[7, 0, 0, 0, 0, 0, 0])])

alarta, 2025-01-13:
line 84: why 41?
my micros serials also start with 13, so changing 11->13 worked

Seems that today and total numbers are off, panels production is not available atm

ArekKubacki, 2025-01-13:
I will prepare the official version tomorrow.
I knew there was something strange about it working for many people. What microinvers do you have? What models are they?

xmatr1x69, 2025-01-13:
> I Increased the time it takes to read data from DTU. I Increased the number of read attempts. Check out version 0.7.1.

Now its working fine on V15.04.09, thank you! 0.7.1 solved it

ArekKubacki, 2025-01-13:
Try 0.7.2 version

alarta, 2025-01-13:
> I will prepare the official version tomorrow. I knew there was something strange about it working for many people. What microinvers do you have? What models are they?

mine are HMT-2250-6T

winbard, 2025-01-13:
Hi,
it seems to work. I see data :)
Thanks for the very good work.

I have
HMT-2000-4T and HMT-2250-6T

winbard, 2025-01-13:
> I have updated DTU firmware but it was days after problem existed.
>
> DTU SN4121A001DE16 Hardware Ver.H09.06.11 Firmware Ver.V00.05.02 And (there was no updates: Micro SN138290601007 Model HMT-2250-6T [...]

Hi,
I see you have update your DTU to V00.05.02.
I have the same Micro and DTU-Pro-S
I install the new firmware but than I have problems to get data from some Micro. Some Micro work.
I get no Power data for e.x. and Alarm that some inputs not there. Not from all micro which not send data.
In Layout was shown die Power data.
I install the 00.02.23 and all work.
Had you problems when you install the v00.05.02?

Sorry for my bad english.

wouterDevelop, 2025-01-13:
> Try 0.7.2 version

This solved for me the issue! Thanks

Golempl, 2025-01-13:
0.7.2 seems starts and entities appears. Can't verify data reporting yet - night and snow.

Jarmo06, 2025-01-16:
Thank you,
Now all working again... 

Core 2025.1.2
Supervisor 2024.12.3
Hoymiles Plant DTU-Pro Sensor v0.7.2

closed 2025-02-07 by ArekKubacki