ArekKubacki/Hoymiles-Plant-DTU-Pro #37: Wrong Measurement of PV Power for "MI" Microinverter

Saarlothri, 2025-01-13:
Since Version 070 the measurements of de PV Power from My Homilies MI Microinverters are fault. They are not zero but they are too small. Round about 1/10 of the right measurement. In 2024 everything was fine.

Hoymiles-.Plant-DTU-Pro
0.70

Core
2025.1.2

Supervisor
2024.12.3

Operating System
14.1

Frontend
20250109.0

ArekKubacki, 2025-01-13:
DTU or OpenDTU?

Saarlothri, 2025-01-13:
It is a Homilies DTU Pro

Saarlothri, 2025-01-13:
The "total... " Entities are not existing any more. 
I haven't seen it because I only used the PV Power to Monitor if everything ist working OK.

Saarlothri, 2025-01-13:
0.71 seems to work ok for me. All PV_Power Entities are there. But I ve seen no total ... Entities.
The only Entitie with the whole Amount of Power is "sensor.hoymiles_pv_pv_power" There is no measurement of Energy.
For me thats OK.
The Reportet Bug that the Measurement of "MI" Devices is fixed! So this Topic can be closed