ArekKubacki/Hoymiles-Plant-DTU-Pro #38: hoymiles unavailable

johanneshelm30, 2025-01-22:
Hi
im a noob so please be patient :)
Everything was working fine until oneday i saw that all my hoymiles data in HA was unavailable. Removed the repositorie from HACS and istalled it again... but now after i add the example configuration to configuration.yaml (changed ip of the dtu and panels number) i get error when i go and check the configuration at "Developer Tools" > " Check & Start" > "Check configuration" The error is-

 Configuration warnings
Platform error 'sensor' from integration 'hoymiles_dtu' - No module named 'pymodbus.pdu.register_read_message'; 'pymodbus.pdu' is not a package

What am i missing or what am i doing wrong?
Is HA dashboard showing data = unavailable if its night and there is no solarpower hence no input from panels?

thanks in advance

[image]
[image]

johanneshelm30, 2025-01-23:
As a noob with some help from Copilot AI i was managed to get bottom of this... reinstalled version 0.62 not sure if it would have worked for latest 0.72 also but seemed like the fix was to install (reinstall) pymodbus...ran the command in termina pip install pymodbus and restared home assistand and voilaa. case closed

johanneshelm30, 2025-01-24:
Trying to get it work in 0.72 version as well... managed to change the

vriezenaar, 2025-02-12:
The v0.8.0 fixed my problems with unavailability of the sensors. So maybe the same for the author? @johanneshelm30? HA 2025.2 broke it for me as well. Very happy you fixed it again @ArekKubacki!