ArekKubacki/Hoymiles-Plant-DTU-Pro #39: HA 2025.2 update

peterophet, 2025-02-06:
I updated to to HA 2025.2, but the integration is not working anymore.
Only me or also other people having this issue?

ArekKubacki, 2025-02-06:
Custom Component version?

peterophet, 2025-02-06:
Latest v0.7.2. No problems for you? Otherwise I get my backup.

ArekKubacki, 2025-02-06:
It looks like for some reason you don't have the latest version of pymodbus
pip install pymodbus

Zeddiboy, 2025-02-06:
I struggle with the same issue after updating to HA 2025.2
forced reinstall of pymodbus did not help, I think I will have to restore for now to previous version.

ArekKubacki, 2025-02-06:
What version did you go from to 2025.2?

I haven't updated yet so I'll be able to check it out after the update.

FrozenHaxor, 2025-02-06:
I confirm this, after updating to 2025.2 the integration is down, updated to whatever the previous latest version was.

marisvs, 2025-02-06:
I have the same problem after the upgrade to HA2025.2 this morning: sensors unavailable and the error message is "No module named 'pymodbus.pdu.register_read_message"

ericdevos, 2025-02-06:
Yes, same issue here, same error message :P

ArekKubacki, 2025-02-06:
In the new HA version they changed pymodbus to version 3.8.3... 
They buried it so deep in the logs... 
Please give me a few days.

ArekKubacki, 2025-02-06:
Please try v0.8.0

peterophet, 2025-02-06:
Up and running again! Great work. Thx!

marisvs, 2025-02-07:
I confirm: after installing v0.8.0 sensors are available again.
Very much appreciated! Thanks!

FrozenHaxor, 2025-02-07:
> Please try v0.8.0

That fixed the issue, thanks.

pml-pml, 2025-02-07:
Great! It works for my HA again! Thanks

ericdevos, 2025-02-07:
Yes, here working again as well after a Update + Reboot!
Thanks for your efforts @ArekKubacki , I will buy you a coffee :)

NB. Do others also have to manually restart the HA server after a HA update?
I find that it often has to do a cold reboot to "find" the DTU-Pro again...

ArekKubacki, 2025-02-07:
It's HACS's fault. It's been like this for a while now.

ArekKubacki, 2025-02-07:
> Yes, here working again as well after a Update + Reboot! Thanks for your efforts [@ArekKubacki](https://github.com/ArekKubacki) , I will buy you a coffee :)
>
> NB. Do others also have to manually [...]

Thanks for coffee :).

closed 2025-02-07 by ArekKubacki