ArekKubacki/Hoymiles-Plant-DTU-Pro #40: Power limiting and ON/OFF control

hugovdkort, 2025-02-08:
Thanks for this great package. This was the only HA "integration" for hoymiles I could get working and it seems you are still very much supporting it and fixing issues. As the title suggests:

Are you planning on adding the write access to the ON/OFF and powerlimiting regiseters or is there a reason why this is not possible?

ArekKubacki, 2025-02-08:
I didn't plan on doing that, but when I dig into the code I'll see what I can do :)

Just tell me why you need to turn off or limit inverters?

hugovdkort, 2025-02-08:
I'd like to be independent of the S-miles app/web interface as much as possible. So for maintenance purposes primarily now.

Later on while having a dynamic energy contract might mean that you have to pay for your solar output (because of low demand/high supply around noon). I'm just planning ahead for when there might come a time when having excess solar is a bad thing...
I'm from the Netherlands by the way and there's been a massive push for solar the last couple years. There's already some companies experimenting with control over domestic systems to help balance the energy grid in peak generation/low load hours. (the people participating get a small fee for helping balance the grid).

not a particularly satisfying answer maybe... I'm definitely not in a rush, but I just saw the hoymiles TCP documentation and it mentioned write control on the on/off and power limit registers. That just made me think.

ArekKubacki, 2025-02-10:
I'll look into the topic.

peterophet, 2025-02-11:
I think this is possible with the openDTU, but could not get it working at my place. So for me this would also be great option.

Anzlinger, 2026-05-02:
Anything new about a solution?