ArekKubacki/Hoymiles-Plant-DTU-Pro #41: Hoymiles unavailabe after last update v0.8.0.

silverrosilverciano, 2025-02-09:
Hi Arek
After last update my entities gone. 

HA docker, core 2025.1.1
Hoymiles HMT-2250

[image]
[image]

biasowicka, 2025-02-10:
I confirm, it also stopped working

ArekKubacki, 2025-02-10:
I kindly ask you to read the information about the changes before clicking. Especially if it is 1 line of text.

[image]

This version is for HA from version 2025.02

silverrosilverciano, 2025-02-10:
You're right. it turned out stupid. i didn't pay attention.

closed 2025-02-10 by silverrosilverciano