ArekKubacki/Hoymiles-Plant-DTU-Pro #43: Error after updating Home Assistant to 2025.9.0

kneschi2, 2025-09-04:
After updating Home Assistant this morning on September 4, 2025, to version 2025.9.0, the DTU plugin no longer starts.
The following error appears in the log file.

Logger: homeassistant.components.sensor
Quelle: helpers/entity_platform.py:451
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 06:35:48 (1 Vorkommnis)
Zuletzt protokolliert: 06:35:48

Error while setting up hoymiles_dtu platform for sensor: Invalid configuration for Hoymiles DTU platform
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 451, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/hoymiles_dtu/sensor.py", line 69, in setup_platform
    raise Exception('Invalid configuration for Hoymiles DTU platform')
Exception: Invalid configuration for Hoymiles DTU platform

Can anyone help fix this?

Regards, Jens

leszegfu, 2025-09-04:
Same error after the upgrade. I've added some logging and found the error below. I'll try to figure it out this morning as I want my PV stats to be up to date.

custom_components/hoymiles_dtu/sensor.py:226
`
ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
`

leszegfu, 2025-09-04:
Here's the fix i applied manually using File Editor in HA. It seems to be working fine now.

/homeassistant/custom_components/hoymiles_dtu/hoymiles/client.py
BEFORE:
`
Line 49:         result = client.read_holding_registers(start_address, count=count, slave=unit_id)
`
AFTER:
`
Line 49:         result = client.read_holding_registers(start_address, count=count, device_id=unit_id)
`

ArekKubacki, 2025-09-04:
Fixed in the new release

leszegfu, 2025-09-04:
Cheers

closed 2025-09-04 by ArekKubacki